xmos/lib_xua #343: xmosdfu does not build on m1 Mac

xross, 2023-09-06:
I made the following change locally to Makefile.OSX64

-       g++ -g -o bin/xmosdfu xmosdfu.cpp -Ilibusb/OSX64 libusb/OSX64/libusb-1.0.0.dylib -Wall
+       g++ -g -o bin/xmosdfu xmosdfu.cpp -Ilibusb/OSX64 libusb/OSX64/libusb-1.0.0.dylib -Wall -arch x86_64 -stdlib=libc++

VergilWang15, 2023-11-17:
@xross Hello Ross, I'm wondering if you are now able to run the DFU utility correctly on the M1 and perform firmware upgrades on the flash memory? It seems that our clients are unable to do so, even though they have successfully compiled the executable file for OSX.

Additionally, could there be some documentation provided for the DFU utility? Attempting to use it across various platforms currently presents some challenges.

xross, 2023-11-17:
@VergilWang15 yes we regression test DFU operation on Windows and MacOS (including intel and arm flavours).

Have you seen this document? https://github.com/xmos/lib_xua/releases/download/v3.5.1/dfu.pdf

xross, 2024-08-16:
This should be resolved by #419 (this build also now CMake based)

closed 2024-08-16 by xross