xmos/lib_xua #345: LRCLK missing while BCLK present

masonpxmos, 2023-09-11:
We got a report from a customer that, their DAC(MAX98357A) is damaged possibly because sometimes XVF3000 outputs BCLK only without outputs LRCLK, According to the datasheet, the LRCLK must provide with BCLK together as follows ".  Do not remove  LRCLK  while  BCLK  is present.  Removing LRCLK while BCLK is present can cause unexpected output behavior including a large DC  output voltage"
Thank Ross for pointing out that lib_xua  will do things like this and miss LRCLK sometimes, Please help to solve this, thanks.

xross, 2023-09-11:
Lib_xua will do this but to the best of my knowledge its has not been an issue in the past. It will occur during a SR change; the LR clock will stop and the bit clock will continue, also at startup, etc.

This is because the bclk is generated in a clock block dividing the master clock using configure_port_clock_output(p_bclk, clk_audio_bclk);

The LR clk is generated via the outputs to the port.

We could reduce this issue by stopping the bclk output whenever we are changing frequency etc. I think stop_clock(clk_audio_bclk) would probably do the trick, but it might be safer to just disable the port output of the clock (configure_port_output())

uvvpavel, 2025-09-16:
Some of my findings after looking at this for a day:
"Just" stopping the bit clock is not sufficient, as it will be reactivated very soon (tens of microseconds), while the left-right clock will be reactivated after the whole board setup, which will take a different time for different boards (currently 15ms for the xk_audio_316_mc).
One possible solution to this would be to move the hardware setup (AudioHwConfig_Mute and AudioHwConfig) before the ports and clocks setup (ConfigAudioPortsWrapper), which does pass the smoke regression, but feels wrong. Not sure what implications it would introduce to set up the DACs without the bit clock running.