xmos/lib_xua #407: Build fail with PDM mics enabled

xross, 2024-07-15:
From user: 

I think I found a bug in sw_usb_audio 8.0.0 and 8.1.0.
In lib_xua/lib_xua/src/core/main.xc,
line 639 (8.0.0)
Line 637 (8.1.0)

it says

#if (XUA_NUM_PDM_MICS > 0)
#endif

but it should be:

#if (XUA_NUM_PDM_MICS > 0)
#if (PDM_TILE == AUDIO_IO_TILE)
, c_ds_output
#endif
, c_pdm_pcm
#endif

I guess.

The bug yields an ‘ error: too few arguments to function `usb_audio_io’
if you try to use PDM mics.
In sw_usb_audio 7.3.1, there is no such bug.

Liu0345, 2024-07-25:
#### I'm also surprised that it seems to have deleted the code for the following image
[image]

xross, 2024-08-13:
This looks to have been erroneously removed in this commit https://github.com/xmos/lib_xua/commit/ca16467158a4ce71118213f7a1f48001b3f29508#diff-1d4846fbc4fb21cffaaf823ad4906e9769013896e17f787697a1cdb6acd302ef

xross, 2024-08-13:
Fixed in #417

PR #417 (2024-08-14): Fix/407
Fixes #407
closed 2024-08-13 by xross