xmos/lib_xua #412: Sending multiple USB device descriptors while using the audio interface

kubapilch, 2024-08-01:
I want to be able to define my own second descriptor to use the XK-AUDIO-316-MC EVK as both an audio device and as a serial COM Port. I have seen from different [example](https://github.com/xmos/lib_xud/tree/develop/examples/AN00124_CDC_VCOM_class) how to achieve solely the COM Port functionality.

My problem is that when using the XUA library, the Endpoint 0 is fully controlled by the `XUA_Endpoint0` function. How can I achieve desired functionality?

xross, 2024-08-01:
Currently you would need to modify lib_xua to achieve this. My recommendation would be to post on www.xcore.com for assistance.

kubapilch, 2024-08-01:
For other people looking for an answer, I am linking the forum [post](https://www.xcore.com/viewtopic.php?t=8815)

kubapilch, 2024-08-01:
@xross Do you know if defining another endpoint i.e. Endpoint 3 as a control endpoint and running another task that will take the code from the XUD code example for sending the CDC device descriptor would work?

xross, 2024-08-02:
Hi @kubapilch 

Im not happy to use the issue system as a general support forum. Please use the xcore.com forum for that - xmos engineers are active there. 

If you would like to raise a specific issue on an enhancement feature allowing more flexible extension of descriptors and addition of endpoints then we'd be happy to accept that.

closed 2024-08-02 by xross