xmos/lib_xua #414: The DFU failed in the v5.27 driver of Thesycon

Liu0345, 2024-08-09:
I used lib xua to release v4.2.0 and test that the DFU was normal（The v5.27 driver for Thesycon）;
[image]
### There may be this problem with the driver DFU in older versions of Thesycon

When I used the https://github.com/xmos/lib_xua/commit/a8dbc8609d4918ce143356a442c561418de24b4c ( latest Merge pull request #404 from shuchitak/dfu 1.1), the DFU failed due to compatibility issues
[image]

Liu0345, 2024-08-09:
Looking forward to testing on #413 at the same time, thank you very much

xross, 2024-08-09:
Please stick to a released version of this lib (ideally the version pinned in sw_usb_audio v8.1.0). There will be driver and documentation updates to follow and a new, tested released of sw_usb_audio.

Liu0345, 2024-08-15:
I tested a phenomenon in the Makefile defined -DUSB TILE=tile[0], DFU is normal.
If you define -DUSB TILE=tile[1] in the Makefile, DFU will fail.

There seems to be some connection with the USB location.

xross, 2025-01-08:
Please re-try with version 9.0.0 of sw_usb_audio and latest DFU scheme

closed 2025-06-27 by xross