xmos/lib_xua #428: XUD_TILE is separate from USB_TILE

henkmuller, 2024-09-23:
It would be nice if XUD_TILE was inherited or linked to the USB_TILE in lib_xud

https://github.com/xmos/lib_xua/blob/d0039cf02b79f353f3a41c3190479ba7871f68b5/lib_xua/api/xua_conf_default.h#L29

xross, 2024-09-23:
I would like to get rid of USB_TILE altogether - it's used such that XUD enables ports on the correct tile. I'd like the lib just to enable ports on the tile it resides on.

ed-xmos, 2025-03-31:
@xross I notice some tests and reboot also use [usb_tile](https://github.com/xmos/lib_xua/blob/d81a486963a8582079a55af62d37c8f41bc0c16e/lib_xua/src/core/support/reboot.xc#L40). Shall I update these to xud_tile also?

xross, 2025-03-31:
> I would like to get rid of USB_TILE altogether - it's used such that XUD enables ports on the correct tile. I'd like the lib just to enable ports on the tile it resides on.

USB_TILE should have been XUD_TILE in this statement

ed-xmos, 2025-03-31:
See https://xmosjira.atlassian.net/wiki/spaces/UAAI/pages/4647682050/Consolidate+USB_TILE+and+XUD_TILE for proposal

closed 2025-06-27 by xross