xmos/lib_xua #449: user_main.h improvement

fabriceo, 2025-01-06:
Hello,
including declartion and cores in lib_xua/main.xc is done via 2 macros defined in user_main.h.
This file name is defined in the OPTIONAL_HEADERS in file lib_xua/module_build_info, and also in LIB_OPTIONAL_HEADERS in lib_xua/lib_build_info.cmake

A better solution, while retro compatible, would be to have separated .h files for the declarations and cores section, instead of basic text written in  #define statements.

I suggest to add the files user_main_declarations.h and user_main_cores.h in the list of optional headers, and to add following piece of code in the main.xc file:
```

#ifdef __user_main_declarations_h_exists__
    #include "user_main_declarations.h"
#endif

    USER_MAIN_DECLARATIONS

    par
    {
#ifdef __user_main_cores_h_exists__
    #include "user_main_cores.h"
#endif
        USER_MAIN_CORES
```

This provide a more flexible and powerfull solution to change the content of main.xc and the original user_main.h solution remains, which is useful for declaring globals or include statements in the top of main.xc

xross, 2025-01-06:
Look like a good suggestion to me, would be happy to accept a PR with this change. 

The documentation would need a related update too, of course.

ed-xmos, 2025-03-28:
I propose to add user_main_globals.h too. The reason is that globals must otherwise be declared elsewhere since adding them to user_main.h via xua_conf.h means they get declared multiple times causing a linker error.

closed 2025-05-23 by xross